srsran/srsRAN_Project #1316: bind_addr ipv6 issue

gustavobsch, 2025-07-30:
I'm trying to connect to AMF over IPv6. I tried different formats in the Yaml, quotes, no quotes, no luck. I also checked all the sample Yamls, couldnt find one with IPv6 configured

```
cu_cp:
  amf:
    addr: amf-ngap.open5gs.ocp.noip.us
    port: 38412
    bind_addr: "fd07:f0d0:2001:14::d"

```

However, gnb always fail to the same error 

```
$ sudo gnb -c config/gnb_n78.yml 

--== srsRAN gNB (commit d8bfdc9d9c) ==--

--bind_addr: Invalid IPV4 address must have four parts (fd07.f0d0.2001.0014.0000.000d)
Run with --help for more information.

--== srsRAN gNB (commit d8bfdc9d9c) ==--

--bind_addr: Invalid IPV4 address must have four parts (fd07:f0d0:2001:14::d)
Run with --help for more information.

--== srsRAN gNB (commit d8bfdc9d9c) ==--

--bind_addr: Invalid IPV4 address must have four parts (fd07:f0d0:2001:0014:0000:000d)
Run with --help for more information.

--bind_addr: Invalid IPV4 address must have four parts (::1)
Run with --help for more information.

```

Please let me know what the right IPv6 format is

Thanks
Gustavo

robertfalkenberg, 2025-08-04:
Although numerous internal modules are already prepared for IPv6, an IPv6-based backhaul is not currently being tested. Please continue to use IPv4 for the time being.

gustavobsch, 2025-12-18:
Any plans to support ipv6 soon?