srsran/srsRAN_Project #1318: srsue is stuck at "Attaching UE" in zmq setup of srsran with open5gs

Dipnarayan, 2025-08-01:
Hello, I am using srsRAN zmq where gnb is connected to AMF on 127.0.0.5:38412
I am running open5gs, gnb and ue in same wsl2

But the srsue is stuck at "Attaching UE"
What can be the issue?

**gnb.yml**

```
cu_cp:
  amf:
    addr: 127.0.0.5                 # The address or hostname of the AMF.
    port: 38412
    bind_addr: 127.0.0.2            # A local IP that the gNB binds to for traffic from the AMF.
    supported_tracking_areas:
      - tac: 1
        plmn_list:
          - plmn: "99970"
            tai_slice_support_list:
              - sst: 1
  inactivity_timer: 7200            # Sets the UE/PDU Session/DRB inactivity timer to 7200 seconds. Supported: [1 - 7200].

ru_sdr:
  device_driver: zmq                # The RF driver name.
  device_args: tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=23.04e6 # Optionally pass arguments to the selected RF driver.
  srate: 23.04                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 75                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 75                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 368500                  # ARFCN of the downlink carrier (center frequency).
  band: 3                           # The NR band.
  channel_bandwidth_MHz: 20         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "99970"                     # PLMN broadcasted by the gNB.
  tac: 1                            # Tracking area code (needs to match the core configuration).
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 12            # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE
  pdsch:
    mcs_table: qam64                # Sets PDSCH MCS to 64 QAM
  pusch:
    mcs_table: qam64                # Sets PUSCH MCS to 64 QAM

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: info                   # Logging level applied to all layers.
  hex_max_size: 0

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.
```

**ue_amq1.conf**

```
[rf]
freq_offset = 0
tx_gain = 50
rx_gain = 40
srate = 23.04e6
nof_antennas = 1

device_name = zmq
device_args = tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6

[rat.eutra]
dl_earfcn = 2850
nof_carriers = 0

[rat.nr]
bands = 3
nof_carriers = 1
max_nof_prb = 106
nof_prb = 106

[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap

[log]
all_level = info
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1

[usim]
mode = soft
algo = milenage
imsi = 999700000000001
k = 0C0A34601D4F07677303652C0462535B
opc = E8ED289DEBA952E4283B54E88E6183CA
imei = 358737060011223

[rrc]
release = 15
ue_category = 4

[nas]
apn = internet
apn_protocol = ipv4

[gw]
netns = ue1
ip_devname = tun_srsue
ip_netmask = 255.255.255.0

[gui]
enable = false
```

ItsGarand, 2025-08-15:
Do you have gnuradio companion running with an appropriate graph?

Dipnarayan, 2025-08-17:
> could you share logs?

Thank you for your help. I have shared the gnb and srsue log below

_dipnarayan@DESKTOP-DG1PQTT:/mnt/c/Users/dipna/Downloads/srsRAN_5G/srsRAN_Project/build/apps/gnb$ sudo ./gnb -c /mnt/c/Users/dipna/Downloads/srsRAN_5G/test1/srsran/gnb.yml_

--== srsRAN gNB (commit d90cd4e26d) ==--

Lower PHY in executor blocking mode.
Available radio types: zmq.
Cell pci=1, bw=20 MHz, 1T1R, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHz

N2: Connection to AMF on 127.0.0.5:38412 completed
==== gNB started ===
Type <h> to view help

_dipnarayan@DESKTOP-DG1PQTT:/mnt/c/Users/dipna/Downloads/srsRAN_5G/srsRAN/build/srsue/src$ sudo ip netns exec ue1 ./srsue ./ue_zmq1.conf_
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins:
Reading configuration file ./ue_zmq1.conf...

Built in Release mode using commit ec29b0c1f on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6
Supported RF device list: UHD zmq file
CHx base_srate=23.04e6
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://127.0.0.1:2000
CH0 tx_port=tcp://127.0.0.1:2001
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...

nicolasbertaud, 2025-09-02:
i  face the same issue . i have trie  in n78, n28, n3 srsure always select n78 FDD

2025-08-27T19:57:22.096548 [PHY-SA ] [I] [    0] Cell search: Setting SSB configuration srate=23.04 MHz; c-freq=**1842.500 MHz**; ss-freq=**1842.050 MHz**; scs=15kHz; pattern=A; **duplex=fdd**;

always on the issue

pgawlowicz, 2025-09-03:
You have to set DL ARFCN and SSB ARFCN:

```
[rat.nr]
bands = 3
nof_carriers = 1
max_nof_prb = 106
nof_prb = 106
dl_nr_arfcn = ....
ssb_nr_arfcn =....
```

pgawlowicz, 2025-09-04:
you have to put the correct values, gNB prints them at start.

JoaoGabrielTN, 2025-09-17:
I am having the same problem in  a multi-ue scenario, the UE logs are empty.

### I did this checklist:

- Is 5GC running? OK
- Does gNB know where AMF is? OK
- Firewall? OK.

- Analyze core and gnb logs CHECK
- Analyze /tmp/gnb.log CHECK
- Analyze core5g terminal CHECK
- Check the sync parameters (plmn and tac). The PLMN configured in 5GC, gnb_zmq.yaml, and the ueX.conf files must be exactly the same. OK
Check the AMF configuration to see if TAC 7 is allowed for PLMN 00101. OK
- Check the UE's credentials OK
- Is the UE registered with the 5GC? The IMSI of each of your UEs (e.g., 001010000000001) needs to be registered in your 5GC's database (usually in the AMF/AUSF), along with their secret keys (K, OPC). OK
- Look for "Authentication Failure" in the AMF logs? OK
- Let's look at the UE log. ARE EMPTY
### UE
``` bash
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ue1_zmq.conf...

Built in Release mode using commit eea87b1 on branch HEAD.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=11.52e6
Supported RF device list: UHD zmq file
CHx base_srate=11.52e6
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://127.0.0.1:2000
CH0 tx_port=tcp://127.0.0.1:2001
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
```
### GNB
``` bash 
--== srsRAN gNB (commit 23b5d8f7b) ==--

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in executor blocking mode.
Available radio types: uhd and zmq.
Cell pci=1, bw=10 MHz, 1T1R, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHz

N2: Connection to AMF on 10.53.1.2:38412 completed
==== gNB started ===
Type <h> to view help
```
I tried to setting the dl_nr_arfcn and the ssb_nr_arfcn parameters and I didn't change anything.

nicolasbertaud, 2025-09-18:
i finally succeed on my side to attach srue zmq to gnb with the following gnb config file :
`ran_node_name: cu_cp_01
gnb_id: 01

cu_cp:
  amf:
    addr: 192.168.50.26                 # The address or hostname of the AMF.
    port: 38412
    bind_addr: 192.168.50.27            # A local IP that the gNB binds to for traffic from the AMF.
    supported_tracking_areas:
      - tac: 1
        plmn_list:
          - plmn: 99970
            tai_slice_support_list:
              - sst: 1
                sd: 0x002116

  inactivity_timer: 7200            # Sets the UE/PDU Session/DRB inactivity timer to 7200 seconds. Supported: [1 - 7200].

ru_sdr:
  device_driver: zmq
  device_args: tx_port=tcp://*:2000,rx_port=tcp://192.168.50.29:2001,base_srate=23.04e6,id=gnb
  srate: 23.04
  tx_gain: 75                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 75                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 362000                  # ARFCN of the downlink carrier (center frequency).
  band: 3                           # The NR band.
  channel_bandwidth_MHz: 20         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: 99970                     # PLMN broadcasted by the gNB.
  tac: 1                            # Tracking area code (needs to match the core configuration).i
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 12            # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 0           # Sets PRACH config to match what is expected by srsUE
  pdsch:
    mcs_table: qam64                # Sets PDSCH MCS to 64 QAM
  pusch:
    mcs_table: qam64                # Sets PUSCH MCS to 64 QAM

log:
  filename: /tmp/gnb.log            # Path of the log file.
    #  all_level: info
  all_level: debug
    #phy_level: debug
    # Logging level applied to all layers.
  hex_max_size: 0

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: true                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.

metrics:
  autostart_stdout_metrics: false       # Optional BOOLEAN (false). Sets whether or note to autostart stdout metrics reporting. Supported [false, true].

  addr: 127.0.0.1                       # Optional TEXT:IPV4 (127.0.0.1). Sets the metrics address. Supported: IPV4 address.
  port: 55555                           # Optional UINT. Sets the metrics UPD port. Supported: [0 - 65535].
  enable_json: true                    # Optional BOOLEAN (false). Enable JSON metrics reporting
  enable_log: false                     # Optional BOOLEAN (false). Enable log metrics reporting
  enable_verbose: false                 # Optional BOOLEAN (false). Enable extended detail metrics reporting

[... 65 lines cut ...]
mode = soft
algo = milenage
opc  = e8ed289deba952e4283b54e88e6183ca
k    = 465b5ce8b199b49faa5f0a2ee238a6bc
imsi = 999700123456780
imei = 353490069873319

[rrc]
release = 15
ue_category = 4

[nas]
apn = internet
apn_protocol = ipv4

[gw]
ip_devname = tun_srsue
ip_netmask = 255.255.255.0
netns = ue1

[gui]
enable = false
`

it was a configuration issue on my side