srsran/srsRAN_Project #1339: No connectivity with VVDN LPRU, UE does not see system information

seppojrantala, 2025-08-27:
Hi,

My O-RAN setup consists of the following:
- Dell PowerEdge XR5610 server running srsRAN software with revision srsRAN 5G gNB version 25.04.0 (d90cd4e26d). The code is compiled as split 7.2 only configuration.
- Falcon-RX/812/G PTP GM synchronizing the CU/DU server and the RU. Falcon-RX has SW: 8.23.1.1-B9
- VVDN LPRU, with System details: 5.4.0-xilinx-v2020.2, HW version: A1-20231206, SW version: 03-v3.0.5 and MAC address: 98:ae:71:04:8c:19 

The srsRAN configuration file is attached. The MTU value of network interface towards LPRU (via Falcon-RX) is set to 9600 and srsran_performance script executed prior to starting the gNB.

Falcon-RX side should be fine, I see no errors, drops or filtered packets in the ports towards the server and LPRU. Both server and LPRU are in sync state. In LPRU, there is Device Status: STABLE and Sync: LOCKED.

Regarding the LPRU configuration, there is the following configuration set:
- Operating band. FOP: 4050.00 MHz, operating Bandwidth: 100 MHz, SCS: 30kHz
- Sync. Domain number: 24, PTP Profile: G_8275_1, Multicast MAC address: Forwardable
- Fronthaul. C/U Plane config: VLAN: 100, DU MAC address: 30:3e:a7:2b:94:39, LLTX-ENDPOINT: eAxC_IDs from 1 to 4: 0, 1, 2, 3 and LLRX-ENDPOINT: eAxC_IDs from 1 to 4: 0, 1, 2, 3
- ORAN. Max PRB per Ethernet frame: 273, Compression method: BFP Compression, IQ width: 9
- RF Calibration. Bandwidth: 100 MHz, ARFCN number: 670000

And finally, the problem is, in shortly as in the title: UE does not see the network. UE is equipped with Keysight Nemo Handy measurement SW which is able to show the signaling messages including the system information broadcasts. No signaling is present between the UE and network. Also the gNB log file is attached. There are several info category "Real-time timing worker woke up late, skipped '2' symbols" prints and one warning also about those. I'm not so familiar with the srsRAN, are those now blocking the connection attempts from the UE?

According to TCP dumps from the server, after starting the gNB there is C/U plane data as Ethernet frames with VLAN:100 tag going from the server to LPRU but not in the other direction. Also the LPRU performance statistics are showing something:
- Throughput. DL: 889.973Mbps, UL: 0.008Mbps

According to the only version of the manual I have regarding the LPRU, publicly available in the internet, there are some scripts in the LPRU to check the traffic:

> 9.3 How to check if RU is getting CU Plane packets from DU?
> By hitting the below mentioned command in the RU console & check for the highlighted messages.
> root@vvdn_lpru: bash /etc/scripts/stat_ru.sh 

But the thing is, I have only the oranuser/oranuser account i.e. non-root user. **Despite several contact attempts towards the VVDN support, they are not willing to provide the root user password. No idea why is that.** As the LPRU is purchased, I think there should be full access to it.

[gnb.log](https://github.com/user-attachments/files/22008018/gnb.log)
[gnb_vvdn_lpru_n77_test_config.yml](https://github.com/user-attachments/files/22008019/gnb_vvdn_lpru_n77_test_config.yml)

mdsahazan, 2025-09-03:
> 01717853529
> What sim card are you using? What is your UE device?
> We have own PLMN in use, the SIM card is from Smartjac. UE was Asus
> Smartphone for Snapdragon insiders. with Keysight Nemo Handy measurement SW.
> But, anyway it does not matter which UE is there since you should see
> something else than noise in R&S handheld spectrum scanner screen when the
> gNB is running.

01717853529

Alan1321, 2025-09-03:
If you are using something like a spectrum analyzer, it should show the SSB transmission from time to time (a peak once in a while). The frequency should be very close to your downlink frequency.

mdsahazan, 2025-09-03:
01717853529

seppojrantala, 2025-09-04:
> 01717853529
> […](#)

Unfortunately that did not work as root user password. I tried also the serial number of my LPRU, did not work either.
This is far too difficult :-/