srsran/srsRAN_Project #1347: srsRAN gNB with NTN issue (RLF issue)

pramod-jonnakuti, 2025-09-09:
Hello team,

We're running **srsRAN gNB with NTN** setup and facing the following issue (**Radio Link Failure**). Because of this, RRC release is happening and UE is getting released. Please find the relevant lines pasted from `gnb.log` file below:

```
2025-09-09T06:46:54.671604 [MAC     ] [I] [   322.6] ue=0: RLF detected. Cause: 100 consecutive HARQ-ACK KOs.
2025-09-09T06:46:54.671628 [DU-MNG  ] [W] ue=0: RLF detected with cause="MAC max KOs reached". Timer of 4000 msec to release UE started...

2025-09-09T06:46:58.670618 [DU-MNG  ] [I] ue=0: RLF timer expired with cause="MAC max KOs reached". Requesting a UE release...
```

Relevant portion of our current srsRAN gnb_config file is as follows:

```
pdsch: # disables HARQ processing
                      
    nof_harqs: 16                           # Optional UNIT (16). Sets the number of Downlink HARQ processes. Supported [2, 4, 6, 8, 10, 12, 16].
    max_nof_harq_retxs: 0
```

pgawlowicz, 2025-09-09:
could you add also the following sections?
```
cu_cp:
  request_pdu_session_timeout: 10
  rrc:
    rrc_procedure_guard_time_ms: 12800 # Set high RRC timeout value.

cell_cfg:
  sched_expert_cfg:
    ta_sched_cfg:
      ta_cmd_offset_threshold: 1
      ta_measurement_slot_period: 40
      ta_measurement_slot_prohibit_period: 300
      ta_target: 0
```

pramod-jonnakuti, 2025-09-10:
Hi @pgawlowicz , we've made the suggested changes but the issue isn't resolved it. We're using the latest srsRAN release by re-building the project. Because of RLF, RRC release is happening and UE is getting released. PFA the `gnb.log` file in compressed format.

[gnb_10_Sep_run1.zip](https://github.com/user-attachments/files/22251153/gnb_10_Sep_run1.zip)

Also, we got parsing error for the below parameter

```
rrc:
    rrc_procedure_guard_time_ms: 12800 # Set high RRC timeout value.

```
so instead we used,

```
rrc:
    rrc_procedure_timeout_ms: 12800 # Set high RRC timeout value.
```

pramod-jonnakuti, 2025-09-10:
Hi @pgawlowicz , as per default config reference (below screenshot), `max_nof_harq_retxs` is already set to `4` in `pusch`. Do we still need to set it explicitly in srsRAN gNB config file?  

[image]

PFA the gNB config file and gnb.log file.

[gnb_config_NTN.yml](https://github.com/user-attachments/files/22253049/gnb_config_NTN.yml)
[gnb_10_Sep_run1.zip](https://github.com/user-attachments/files/22253048/gnb_10_Sep_run1.zip)

For NTN setup, we're using TM500 UE simulator that has NTN channel emulator that emulates NTN link delay.

mdsahazan, 2025-09-15:
> Hi @pgawlowicz <https://github.com/pgawlowicz> , sure. PFA the ngap and
> mac pcap files below (along with gnb.log file):
>
> logs_mac_ngap.zip

01717853529

pramod-jonnakuti, 2025-09-15:
@pgawlowicz , Please find the UE coordinates below:

```
UE XYZ Coordinates:
x = 1328347
y = 6076983
z = 1408943

gNB  and satellite XYZ Coordinates:
x = 24184355 
y = 34538839 
z = 0
```

Arun-Cdot, 2025-09-16:
@pgawlowicz I’m also working with Pramod on this. I think the gNB uses a timer to wait for HARQ ACKs. For terrestrial networks this timer is probably set quite small, but for 5G GEO NTN it needs to be much larger. The issue of 'RLF detected. Cause: 100 consecutive HARQ-ACK KOs' might be related to this.

I also observed that this issue shows up when the downlink data rate is set to 1 Mbps, but not when it is set to 13.6 kbps. Both runs were stopped after 10 seconds of data transmission. In the low data rate case, I noticed multiple PUCCH detections between each PDSCH transmission (whether using 256QAM or 64QAM), whereas this was not the case at the higher data rate. This could be the ACKs from the UE.

I’m attaching the gNB logs for your review.

[gnb_1Mbps.log](https://github.com/user-attachments/files/22356701/gnb_1Mbps.log)

[gnb_13p6kbps.log](https://github.com/user-attachments/files/22356705/gnb_13p6kbps.log)

pgawlowicz, 2025-10-24:
Hi @pramod-jonnakuti and @Arun-Cdot, could you pull the newest code version and try again? 
The relevant timeouts were extended to take NTN link RTT into account.

Arun-Cdot, 2025-11-18:
Hi @pgawlowicz ,

We conducted multiple test runs while attempting to transmit 20 Mbps to the TM500 UE. The observations are as follows:

With harq_mode_b = false and nof_harqs = 16, the data rate is approximately 4.5 Mbps.

With harq_mode_b = false and nof_harqs = 8, the data rate is approximately 2.2 Mbps.

With harq_mode_b = false, nof_harqs = 8, and the RTT reduced to **667 ns (from 240 ms)**, the data rate remains approximately 2.2 Mbps, indicating that the RTT is not affecting the throughput.

We are unable to set nof_harqs = 32 because the TM500 does not support this configuration.

With harq_mode_b = true, the data rate becomes unstable and fluctuates between 12 Mbps and 18 Mbps.

Additionally, when harq_mode_b = true, the logs show messages such as:
[SCHED] [I] [922.6] rnti=0x4601 h_id=4: Discarding DL HARQ process TB with tbs=1377. Cause: Maximum number of reTxs 0 exceeded.

pgawlowicz, 2025-11-20:
Hi @Arun-Cdot,

The fluctuations might be related to RLC reordering timeouts. Could you run with RLC UM?
You can probably run with this extra config file that changes the default RLC mode for five_qi: 9 (if this is a bearer id you use).

[qos.yml](https://github.com/user-attachments/files/23650112/qos.yml)

Best regards,
Piotr

mdsahazan, 2025-11-20:
> Please find attached the config file (extracted from the log) and the
> complete log file. We haven’t yet tested the changes you suggested. We’ll
> add harq_mode_b: true, run the test, and let you know if it resolves the
> config_file_extracted_from_log_file.txt
> gnb_NTN_new.zip

01806647397

01717853529