srsran/srsRAN_Project #1361: Grafana metrics is no longer working

hadziarmin, 2025-09-17:
I took over the new 25.04 release and metrics in grafana are not working anymore. First in the new release there are no config parameters metrics.addr and metrics.port and the reference configuration is not updated. I started digging to figure out how it works then and I guess somehow over sockets. Anyway, after solving the internet access problem for grafana container i run into next one:

grafana  | logger=context userId=0 orgId=1 uname= t=2025-09-17T08:13:32.682018558Z level=error msg="Request Completed" method=POST path=/api/ds/query status=500 remote_addr=172.19.1.1 time_ms=15 duration=15.837926ms size=103 referer="http://localhost:3300/d/a8a69f8a-1dd1-4496-8f3f-33de8988a823/srsran-metrics?from=now-5m&orgId=1&refresh=1s&timezone=browser&to=now" handler=/api/ds/query status_source=downstream errorReason=InternalError errorMessageID=plugin.downstreamError error="client: failed to query data: missing organization in datasource configuration"

I noticed that versions of InfluxDb and grafana are changed in the docker/.env file and telegraf is added. 
The grafana metrics helps a lot. Can someone help to enable it again?

metrics:
  enable_json: true       # Enable reporting metrics in JSON format
  #addr: 172.19.1.4                # Metrics-server IP
  #port: 55555                     # Metrics-server Port
  enable_log: false                     # Optional BOOLEAN (false). Enable log metrics reporting
  enable_verbose: false

hadziarmin, 2025-10-02:
I did, but as in new issues is explained and here, the parameters addr and port are no longer used in metrics. I tried to specify them under remote_control command, but this seems to start remote_control server to listen and metrics are still not working.

robertfalkenberg, 2025-10-02:
Perhaps this can help

https://github.com/srsran/srsRAN_Project/issues/1363#issuecomment-3360426374

nicolasbertaud, 2025-10-07:
i undertand in the release there is a change to expose the metrics in websocket . i finally succeed to activate metrics with following structure in yml  .          
`metrics:
  enable_json: true
remote_control:
  bind_addr: 127.0.0.1
  port: 55555
  enabled: true

metrics:
  layers:
    enable_app_usage: true            #     Enable application usage metrics
    enable_ngap: true                      #          Enable NGAP metrics
    enable_rrc: true                         #      Enable CU-CP RRC metrics
    enable_e1ap: true                     #    Enable E1AP metrics
    enable_pdcp: true                     #      Enable PDCP metrics
    enable_cu_up_executor: true #  Whether to log CU-UP executor metrics
    enable_sched: true                   # Enable DU scheduler metrics
    enable_rlc: true                         #    Enable RLC metrics
    enable_mac: true                      #   Enable MAC metrics
    enable_executor: true              #  Whether to log DU-high executor metrics
    enable_du_low: true                #   Enable DU low metrics (upper physical layer)
    enable_ru: true                         #  Enable Radio Unit metrics
  periodicity:
    app_usage_report_period: 1000 #    Application resource usage metrics report period (in milliseconds)
    cu_cp_report_period: 1000          #    CU-CP metrics report period in milliseconds
    cu_up_report_period: 1000         #    CU-UP metrics report period in milliseconds
    du_report_period:  1000              #   DU statistics report period in milliseconds`

it works with the following load 
git branch -vv
* main 11c9bbabb6 [origin/main] Update main
nicolas@ran02:~/srsRAN_Project$ git show -s --format="%ci" 11c9bbabb6
2025-09-15 07:54:53 +0000

unfortunately with the latest load i don't suceed to make it working with same set up
 git branch -vv
* main cdc93a6092 [origin/main] Update main
nicolas@ran01:~/srsRAN_Project$ git show -s --format="%ci" cdc93a6092
2025-09-29 08:04:10 +0000

seems metrics are not activated  . Is there a new change ?