srsran/srsRAN_Project #1363: regression with  metrics

nicolasbertaud, 2025-09-18:
hi
i report an issue we are facing with latest build . it i snot ^possible to activate the metrics anymore

it was possible to activate metrics in that build
nicolas@ran01:~/srsRAN_Project/build$ git branch -vv
* main d90cd4e26d [origin/main] Update main
nicolas@ran01:~/srsRAN_Project/build$ git show -s --format="%ci" d90cd4e26d
2025-07-28 15:56:50 +0000

it is not working anymore in the build
nicolas@ran02:~/srsRAN_Project/configs$ git branch -vv
* main 11c9bbabb6 [origin/main] Update main
nicolas@ran02:~/srsRAN_Project/configs$ git show -s --format="%ci" 11c9bbabb6
2025-09-15 07:54:53 +0000
nicolas@ran02:~/srsRAN_Project/config

all metrics parameters doesn't seems to be recognized

fisrt i face the  issue with metrics.addr
sudo /home/nicolas/srsRAN_Project/build/apps/gnb/gnb  -c /home/nicolas/srsRAN_Project/configs/gnb_zmq_n65.yml

--== srsRAN gNB (commit 11c9bbabb6) ==--

INI was not able to parse metrics.addr
Run with --help for more information.

I comment it in yml and then i face issue with port 

sudo /home/nicolas/srsRAN_Project/build/apps/gnb/gnb  -c /home/nicolas/srsRAN_Project/configs/gnb_zmq_FDDonly_B65.yml

--== srsRAN gNB (commit 11c9bbabb6) ==--

INI was not able to parse metrics.port
Run with --help for more information.

Can you clarify the status of metrices with latest build ?

nicolasbertaud, 2025-09-22:
my configuration the metrics part is commented as it is not working anymore  `sudo cat  gnb_zmq_FDDonly_B65_ran02.yml
# This configuration file example shows how to configure the srsRAN Project gNB to allow srsUE to connect to it.
# This specific example uses ZMQ in place of a USRP for the RF-frontend, and creates an FDD cell with 10 MHz bandwidth.
# To run the srsRAN Project gNB with this config, use the following command:
#   sudo ./gnb -c gnb_zmq.yaml
ran_node_name: ran_02
gnb_id: 02

cu_cp:
  amf:
    addr: 192.168.50.26
    port: 38412
    bind_addr: 192.168.50.32
    supported_tracking_areas:
      - tac: 2
        plmn_list:
          - plmn: 99970
            tai_slice_support_list:
              - sst: 1
                sd: 0x000000

  inactivity_timer: 7200
  rrc:                                # <-- déplacer ici
    #    rrc_procedure_timeout_ms: 10000  # Timeout RRC augmenté à 10s

ru_sdr:
  device_driver: zmq
  device_args: tx_port=tcp://192.168.50.32:2002,rx_port=tcp://192.168.50.29:2003,base_srate=23.04e6,id=gnb2
  srate: 23.04
  tx_gain: 75
  rx_gain: 75

cell_cfg:
  dl_arfcn: 432000
  band: 65
  channel_bandwidth_MHz: 20
  common_scs: 15
  plmn: 99970
  tac: 2
  pci: 2
  pdcch:
    common:
      ss0_index: 0
      coreset0_index: 12
    dedicated:
      ss2_type: common
      dci_format_0_1_and_1_1: false
  prach:
    prach_config_index: 2
  pdsch:
    mcs_table: qam64
  pusch:
    mcs_table: qam64

log:
  filename: /tmp/gnb.log
  all_level: debug
  hex_max_size: 0

pcap:
  mac_enable: false
  mac_filename: /tmp/gnb_mac.pcap
  ngap_enable: true
  ngap_filename: /tmp/gnb_ngap.pcap

    #metrics:
  #  autostart_stdout_metrics: false       # Optional BOOLEAN (false). Sets whether or note to autostart stdout metrics reporting. Supported [false, true].

    #  addr: 127.0.0.1                       # Optional TEXT:IPV4 (127.0.0.1). Sets the metrics address. Supported: IPV4 address.
    #port: 55555                           # Optional UINT. Sets the metrics UPD port. Supported: [0 - 65535].
    #enable_json: true                    # Optional BOOLEAN (false). Enable JSON metrics reporting
    #enable_log: false                     # Optional BOOLEAN (false). Enable log metrics reporting
    #enable_verbose: false                 # Optional BOOLEAN (false). Enable extended detail metrics reporting

    #  layers:
    #    enable_app_usage: true               # Optional BOOLEAN (false). Enable application usage metrics
    #    enable_sched: false                   # Optional BOOLEAN (true). Enable DU scheduler metrics
    #    enable_rlc: true                     # Optional BOOLEAN (false). Enable RLC metrics
    #    enable_mac: true                     # Optional BOOLEAN (false). Enable MAC metrics
    #    enable_executor: false                # Optional BOOLEAN (false). Whether to log DU-high executor metrics
    #    enable_du_low: false                  # Optional BOOLEAN (false). Enable DU low metrics (upper physical layer)
    #    enable_ru: false                      # Optional BOOLEAN (false). Enable Radio Unit metrics

    #  periodicity:
    #    app_usage_report_period: 1000         # Optional UINT (1000). Application resource usage metrics report period in milliseconds
    #    du_report_period: 1000                # Optional UINT:INT (1000). DU statistics report period in milliseconds. Format: [0 - 10485760].

      #expert_execution:
      #  threads:
      #   non_rt:
      #      nof_non_rt_threads: 2
`

nicolasbertaud, 2025-10-01:
i  understand nothing to your answer 
someone tells me   "The addr and port have been removed from the metrics subcommand. You can verify this by running your/repo/build/path/apps/gnb/gnb metrics --help, which should show:

--== srsRAN gNB (commit c11773fad6) ==--

Metrics configuration
Usage: ./apps/gnb/gnb metrics [OPTIONS] [SUBCOMMAND]

Options:
  -h,--help                   Print this help message and exit
  --autostart_stdout_metrics BOOLEAN [false]
                              Autostart stdout metrics reporting
  --enable_json               Enable JSON metrics reporting
  --enable_log                Enable log metrics reporting
  --enable_verbose            Enable extended detail metrics reporting

Subcommands:
  layers                      Layer basis metrics configuration
  periodicity                 Metrics periodicity configuration  

We redesigned the metrics framework to use websockets. You can find its parameters under the subcommand remote_control"

i saw some parameters are removed  --> this is ok 
i also saw new sub_command is not there !!!!

The sub command remote_control  exits  but how can we use it if it is not recognize  " INI was not able to parse metrics.remote_control.++"

i don't think explanation are consistent . what is wrong in the config file 

:wq:
`metrics:
  autostart_stdout_metrics: false       # Optional BOOLEAN (false). Sets whether or note to autostart stdout metrics reporting. Supported [false, true].

  enable_json: true                    # Optional BOOLEAN (false). Enable JSON metrics reporting
  #enable_log: false                     # Optional BOOLEAN (false). Enable log metrics reporting
  #enable_verbose: false                 # Optional BOOLEAN (false). Enable extended detail metrics reporting

  layers:
    enable_app_usage: true            #     Enable application usage metrics
    enable_ngap: true                      #          Enable NGAP metrics
    enable_rrc: true                         #      Enable CU-CP RRC metrics
    enable_e1ap: true                     #    Enable E1AP metrics
    enable_pdcp: true                     #      Enable PDCP metrics
    enable_cu_up_executor: true #  Whether to log CU-UP executor metrics
    enable_sched: true                   # Enable DU scheduler metrics
    enable_rlc: true                         #    Enable RLC metrics
    enable_mac: true                      #   Enable MAC metrics
    enable_executor: true              #  Whether to log DU-high executor metrics
    enable_du_low: true                #   Enable DU low metrics (upper physical layer)
    enable_ru: true                         #  Enable Radio Unit metrics
    periodicity:
      app_usage_report_period: 1000 #    Application resource usage metrics report period (in milliseconds)
      cu_cp_report_period: 1000          #    CU-CP metrics report period in milliseconds
      cu_up_report_period: 1000         #    CU-UP metrics report period in milliseconds
      du_report_period:  1000              #   DU statistics report period in milliseconds
    remote_control:
      enabled: true                   #           Enables the Remote Control Server
      bind_addr: "127.0.0.1"   #           Remote Control Server bind address
      port: 55555        `

sudo /usr/local/bin/gnb -c  /home/nicolas/srsRAN_Project/configs/gnb.yml

--== srsRAN gNB (commit 11c9bbabb6) ==--

INI was not able to parse metrics.remote_control.++
Run with --help for more information.
nicolas@ran02:~/srsRAN_Project/configs$

robertfalkenberg, 2025-10-02:
Hi @nicolasbertaud,

the `port` and `addr` of the `metrics` subcommand have been removed, because the UDP-based metrics have been replaced recently with a WebSocket approach. The public docs are not updated yet but will be so soon.

Until then, enable remote control and json metrics as follows:

```yml
metrics:
  enable_json: true
remote_control:
  bind_addr: 0.0.0.0
  enabled: true
```
Configure layers and periodicity as before, e.g.:
```yml
metrics:
  layers:
    enable_app_usage: true
    enable_mac: true
  periodicity:
    app_usage_report_period: 10000
    cu_cp_report_period: 1000
    cu_up_report_period: 1000
    du_report_period: 1000
```

The socket listens on port `8001` and metrics can be subscribed by sending the following json command:
```json
{
   "cmd" : "metrics_subscribe"
}
```
There is also a ``"metrics_unsubscribe"`` command in case you want to unsubscribe.

Here is a simple python example that prints the json metrics to stdout:
```python
#!/usr/bin/env python3

from contextlib import suppress
import os
import json
from time import sleep
import websocket

def _on_open(ws: websocket.WebSocketApp):
    ws.send(json.dumps({"cmd": "metrics_subscribe"}))

def _on_message(_ws: websocket.WebSocketApp, message: str):
    with suppress(json.JSONDecodeError):
        metric = json.loads(message)
        if "cmd" not in metric:
            print(json.dumps(metric))

if __name__ == "__main__":
    ws_app = websocket.WebSocketApp(
        "ws://" + os.environ["WS_URL"],
        on_open=_on_open,
        on_message=_on_message,
    )
    while ws_app.run_forever():  # Returns False when the connection is closed
        sleep(1)
```
Run with
```bash
WS_URL="127.0.0.1:8001" ./ws_adapter.py
```