srsran/srsRAN_Project #1372: UE does not see gNB

luyuanfan, 2025-09-29:
## Problem Description

I've been trying to set up a gNB with srsRAN, Open5GS, and a X410 USRP device. However, the network does not show up in my UE's list of networks. I'm think that the problem could be in the configuration files. I'm also not sure if the way I connect the antenna could cause the problem.

## Setup Details

Core Platform: Open5GS 
RF device: USRP X410 (connected via ethernet cable with ip address set to `192.168.20.2`)
UHD version: 4.8.0
OS: Ubuntu 24.04.1 LTS
UE: Nothing phone 2

## Attached Files

- [open5gs-config.txt](https://github.com/user-attachments/files/22603903/open5gs-config.txt)
- [gnb-config.txt](https://github.com/user-attachments/files/22603909/gnb-config.txt)
- [open5gs-log.txt](https://github.com/user-attachments/files/22603924/open5gs-log.txt)
- [gnb-log.txt](https://github.com/user-attachments/files/22603928/gnb-log.txt)
- [image]

andrepuschmann, 2025-09-30:
Hey - if you see constant lates or underflows it's an indication for network config issues, perhaps rx/tx buffering on the NIC, kernel?
Please run `./benchmark_rate --rx_rate 30.72e6 --tx_rate 30.72e6`  and see if that works without issues on your machine.  If it does, try `./benchmark_rate --rx_rate 122.88e6 --tx_rate 122.88e6` as well.

luyuanfan, 2025-09-30:
> Hey - if you see constant lates or underflows it's an indication for network config issues, perhaps rx/tx buffering on the NIC, kernel? Please run `./benchmark_rate --rx_rate 30.72e6 --tx_rate [...]

Hi! Thank you for the response. I did forget to run the srsran performance tuning script (for network buffer tuning). After running that script and your suggestion, it seems to have fixed the underflow problem. However, the UE still does not see the network.

I don't see much warning when I set the gnb `log: all_level:` to warning, so I'm taking that as there's no network config issues. So I'm guessing that there might still be problems with the configurations. Is there anything that might be problematic in what I have?

robertfalkenberg, 2025-10-01:
I've seen your PLMN is 999/70. Unless you are 100% confident the UE supports that network (e.g. you successfully tested the same UE + PLMN with different equipment), I suggest you configure the network with Test-PLMN 001/01.

luyuanfan, 2025-10-02:
> I've seen your PLMN is 999/70. Unless you are 100% confident the UE supports that network (e.g. you successfully tested the same UE + PLMN with different equipment), I suggest you configure the network with Test-PLMN 001/01.
> 
> Further details in [#100 (reply in thread)](https://github.com/srsran/srsRAN_Project/discussions/100#discussioncomment-6496164)

Hi - Yes I believe I have tried this combination. My UE would not allow me to add an APN 00101. However, even if this is true, the UE should still be able to see the network. I've tried many combinations including this but it doesn't work.

robertfalkenberg, 2025-10-02:
Another example is found here:

https://github.com/srsran/srsRAN_Project/issues/1348

luyuanfan, 2025-10-17:
An update: frankly someone just told me that b200 or b210 has worse hardware in general (they are good enough for LTE), so something like x310 or x410 is recommended for gNB.

closed 2025-10-17 by luyuanfan