srsran/srsRAN_Project #1375: UE no longer connects with latest srsRAN 5G (Benetel RU v1.0.4) — works on May build

ismagom, 2025-09-29:
### Discussed in https://github.com/srsran/srsRAN_Project/discussions/1374

<div type='discussions-op-text'>

<sup>Originally posted by **ujjwalpawar** September 29, 2025</sup>
Hi team,

After updating to the latest `srsRAN_5G` (current `main` as of this week), our UE no longer connects. The exact same setup and config work reliably on a build from May.

**RU / split:** Benetel RU v1.0.4 (O-RAN 7.2)  
**DU/CU:** srsRAN 5G (RU OFH + gnb), standalone core  
**OS:** Ubuntu 22.04  
**Bandwidth / band:** n77, 100 MHz, SCS 30 kHz  
**Antennas:** 4x4 MIMO  
**Fronthaul:** DPDK (EAL args below)
UE can see the cell but never does the RACH successfully (no uplink activity at RU side)
On the latest build, RU-side counters (e.g., RX_TOTAL / RX_ON_TIME) are consistently lower than on the May build under configuration
### UE logs
[image]

### RU logs with latest version
[image]

###  RU logs with may version
[image]

## Configuration (works on May build, fails on latest)

```yaml
cu_cp:
  amf:
    addr: 172.30.129.35
    port: 38412
    bind_addr: 10.0.0.3
    supported_tracking_areas:
      - tac: 1
        plmn_list:
          - plmn: "99999"
            tai_slice_support_list:
              - sst: 1

ru_ofh:
  t1a_max_cp_dl: 535
  t1a_min_cp_dl: 286
  t1a_max_cp_ul: 535
  t1a_min_cp_ul: 286
  t1a_max_up: 390
  t1a_min_up: 80
  ta4_max: 500
  ta4_min: 25
  is_prach_cp_enabled: false
  compr_method_ul: bfp
  compr_bitwidth_ul: 9
  compr_method_dl: bfp
  compr_bitwidth_dl: 9
  compr_method_prach: bfp
  compr_bitwidth_prach: 9
  enable_ul_static_compr_hdr: true
  enable_dl_static_compr_hdr: true
  iq_scaling: 5.5
  cells:
    - network_interface: 0000:86:01.0
      ru_mac_addr: 70:b3:d5:e1:5f:e9
      du_mac_addr: 00:11:22:33:77:88
      vlan_tag_cp: 11
      vlan_tag_up: 11
      prach_port_id: [4,5,6,7]
      dl_port_id: [0, 1,2,3]
      ul_port_id: [0, 1,2,3]

cell_cfg:
  dl_arfcn: 660000
  band: 77
  channel_bandwidth_MHz: 100
  common_scs: 30
  plmn: "99999"
  tac: 1
  pci: 11
  nof_antennas_dl: 4
  nof_antennas_ul: 4
  prach:
    prach_config_index: 7
    prach_root_sequence_index: 11
    zero_correlation_zone: 0
    prach_frequency_start: 0
    preamble_trans_max: 4
  ssb:
    ssb_block_power_dbm: 0
  tdd_ul_dl_cfg:
    dl_ul_tx_period: 10
    nof_dl_slots: 7
    nof_dl_symbols: 6
    nof_ul_slots: 2
    nof_ul_symbols: 4

hal:
  eal_args: "--lcores (0-1)@(2-55) -a 0000:86:01.0"

expert_phy:
  allow_request_on_empty_uplink_slot: false

log:
  filename: stdout
  all_level: warning
  config_level: debug

pcap:
  mac_enable: false
  mac_filename: /tmp/gnb_mac.pcap
  ngap_enable: false
  ngap_filename: /tmp/gnb_ngap.pcap
  
  
</div>

sauka, 2025-09-30:
Hi @ujjwalpawar, 

I see you are using the long format PRACH. Could you check the `lf_prach_slot_id` parameter in Benetel's ru_config file? It should be set to 0. In the old RU FW releases this parameter didn't exist but the behaviour of the RU was corresponding to lf_prach_slot_id=1. In the new FW releases the default value is lf_prach_slot_id=0, and latest srsRAN expects this configuration too.

ujjwalpawar, 2025-10-01:
Is there any configuration parameter i can try to fix this. Or any change to the code?

sauka, 2025-10-01:
There is no config option to force an old behaviour. Could you upgrade the FW of your Betenel RU?

ujjwalpawar, 2025-10-01:
Only if it is the last option. Can you point me to the code where this is handled?

sauka, 2025-10-01:
I would recommend to upgrade the FW as Benetel fixed a few important issues since the version 1.0.4. If it it is not an option, you could try reverting this commit https://github.com/srsran/srsRAN_Project/commit/c427ebf7. Then you will be able to use `enable_lf_prach_slot_autoderivation` param in ru_ofh section of the config file. Setting it to `true` will force an old behaviour. However it is non standard - that is why we dropped it.

ujjwalpawar, 2025-10-01:
Using short prach format won't create this issue i suppose. But never got short prach to work with benetel I have tried this before.
```
  prach:
    prach_config_index: 159
    prach_root_sequence_index: 1
    zero_correlation_zone: 0
    prach_frequency_start: 0
    preamble_trans_max: 4
```

Do have any suggestion for short prach ?

sauka, 2025-10-02:
> Using short prach format won't create this issue i suppose. But never got short prach to work with benetel I have tried this before.
>
> ```
>   prach:
>     prach_config_index: 159 [...]

I don't have the complete ru_config on hand for v1.0.4. For the latest Benetel FW release you can find it here: https://docs.srsran.com/projects/project/en/latest/tutorials/source/oranRU/source/rus/r550.html , the ru config file in the tutorial specifically targets the short PRACH. 

Basically you need to set `prach_format=short` and `compression=dynamic_compressed` in the RU.

On the gNB side you need these OFH settings (I am not including timing window and other params):
```
ru_ofh:
  is_prach_cp_enabled: true
  compr_method_ul: bfp
  compr_bitwidth_ul: 9
  compr_method_dl: bfp
  compr_bitwidth_dl: 9
  compr_method_prach: bfp
  compr_bitwidth_prach: 9
  enable_ul_static_compr_hdr: false
  enable_dl_static_compr_hdr: false
```
And the PRACH parameters in the cell config:
```
cell_cfg:
  prach:
    prach_config_index: 159
    prach_root_sequence_index: 1
    zero_correlation_zone: 0
```

Let me know please if you have any further questions

ujjwalpawar, 2025-10-02:
Sadly, it is not working with short prach in our setup. 
I am using latest build. Here is my config. I don't see any error/warning. Do you see anything wrong with the config? 
### ru_config 
```

mimo_mode=1_2_3_4_4x4

downlink_scaling=0

prach_format=short

compression=dynamic_compressed

lf_prach_compression_enable=false

cplane_per_symbol_workaround=disabled

cuplane_dl_coupling_sectionID=disabled

flexran_prach_workaround=disabled

dl_ul_tuning_special_slot=0xfd00000
```
gnb_config

```
  is_prach_cp_enabled: true
  compr_method_ul: bfp
  compr_bitwidth_ul: 9
  compr_method_dl: bfp
  compr_bitwidth_dl: 9
  compr_method_prach: bfp
  compr_bitwidth_prach: 9
  enable_ul_static_compr_hdr: false
  enable_dl_static_compr_hdr: false
  iq_scaling: 20
```

```
cell_cfg:
  dl_arfcn: 660000
  band: 77
  channel_bandwidth_MHz: 100
  common_scs: 30
  plmn: "99999"
  tac: 1
  pci: 11
  nof_antennas_dl: 4
  nof_antennas_ul: 4
  prach:
    prach_config_index: 152
    prach_root_sequence_index: 1
    zero_correlation_zone: 0
    #prach_frequency_start: 0
    #preamble_trans_max: 4
#    ports: 1
  ssb:
    ssb_block_power_dbm: 0
  tdd_ul_dl_cfg:
    dl_ul_tx_period: 10
    nof_dl_slots: 7
    nof_dl_symbols: 6
    nof_ul_slots: 2
    nof_ul_symbols: 4
```

debashish216, 2025-12-28:
@ismagom  Thank you.
My RAN550 firmware version is v1.4.1. Do you think I need this patch for newer firmware as well?

ujjwalpawar, 2025-12-28:
No this issue is with firmware 1.0.4